openenergymonitor/emonhub #37: Errors ? due to date

mikechristiansen, 2017-08-01:
Errors as below for 24 hours on the 31/Aug/17 starting around 06:45 NZT and failed to work after restarts until the next day. 

2017-07-31 06:57:56,449 INFO SMASolar Log into the SMA solar inverter 00:80:25:2F:B4:28
2017-07-31 06:57:56,451 INFO SMASolar Opening bluetooth address 00:80:25:2F:B4:28
2017-07-31 06:57:58,067 DEBUG SMASolar initaliseSMAConnection
2017-07-31 06:57:58,516 DEBUG SMASolar packet count = 0101
2017-07-31 06:57:58,587 DEBUG SMASolar packet count = 0102
2017-07-31 06:57:58,599 DEBUG SMASolar logon
2017-07-31 06:57:58,731 DEBUG SMASolar packet count = 0103
2017-07-31 06:57:58,824 ERROR SMASolar Failed to grab all the bytes needed for a Level 2 packet
2017-07-31 06:57:58,877 ERROR SMASolar [‘Traceback (most recent call last):\n’, ’ File “/home/pi/emonhub/src/interfacers/EmonHubSMASolarInterfacer.py”, line 203, in read\n self._login_inverter()\n’, ’ File “/home/pi/emonhub/src/interfacers/EmonHubSMASolarInterfacer.py”, line 90, in _login_inverter\n dictInverterData = SMASolar_library.getInverterDetails(self._btSocket, self._packet_send_counter, self.mylocalBTAddress, self.MySerialNumber)\n’, ’ File “/home/pi/emonhub/src/smalibrary/SMASolar_library.py”, line 141, in getInverterDetails\n data=request_data(btSocket, packet_send_counter, mylocalBTAddress, MySerialNumber, 0x58000200, 0x00821E00, 0x008220FF)\n’, ’ File “/home/pi/emonhub/src/smalibrary/SMASolar_library.py”, line 322, in request_data\n bluetoothbuffer = read_SMA_BT_Packet(btSocket, packet_send_counter, True,mylocalBTAddress)\n’, ’ File “/home/pi/emonhub/src/smalibrary/SMASolar_library.py”, line 81, in read_SMA_BT_Packet\n raise Exception(“Failed to grab all the bytes needed for a Level 2 packet”)\n’, ‘Exception: Failed to grab all the bytes needed for a Level 2 packet\n’]
2017-07-31 06:58:02,502 INFO SMASolar Log into the SMA solar inverter 00:80:25:2F:B4:28
2017-07-31 06:58:02,504 INFO SMASolar Opening bluetooth address 00:80:25:2F:B4:28
2017-07-31 06:58:02,506 ERROR SMASolar (16, ‘Device or resource busy’)
2017-07-31 06:58:02,518 ERROR SMASolar Bluetooth error while connecting to 00:80:25:2F:B4:28

SMASolar_library.getInverterDetails may be the culprit? Might be a date bug in reporting from inverter? 

See https://community.openenergymonitor.org/t/sma-solar-pv-inverters-monitoring-over-bluetooth/3383/77?u=mikec for details. 

Fantastic software developers! Keep up the amazing work :)

glynhudson, 2017-08-02:
Thanks for reporting the issue. @stuartpittaway worked on the SMA integration. Hopefully he will be able to take a look.

stuartpittaway, 2017-08-14:
This issue definitely appears to be date related suspect one of the decoding routines is incorrectly parsing a value in the date as a special escaped byte and then incorrectly decoding the packet.

stuartpittaway, 2018-02-10:
If issue appears to be occurring again today (10 Feb 2018) so must be something to do with the date and decoding escape characters from the inverter.

stuartpittaway, 2018-02-13:
As expected, the SMA inverters use the 0x7e byte for some escape characters - both of these dates have 0x7E as the second byte.

Monday, 31 July 2017 08:00:00
1501488000
597EE380

Saturday, 10 February 2018 08:00:00
1518249600
5A7EA680

stuartpittaway, 2018-08-05:
Actually mine stopped working again today, needs investigate!