openenergymonitor/emonhub #45: single NaN result from any register blocks all subsequent uploads to emoncms

yekorzh, 2018-01-17:
I'm using this interface to communicate with Fronius Inverter and noticed that sometimes on rare occasion, result read from the register can be NaN. This "NaN" then goes into the databuffer, gets converted to JSON and later rejected by emoncms as NaN is not part of JSON spec. Data sample with NaN result stays in the databuffer blocking all other valid samples - it gets rejected by emoncms again and again. Restarting emonhub resolves it.

Some error checking needs to be implemented to prevent this. I suggest that the whole data sample needs to be discarded if any of the values are invalid.

glynhudson, 2018-01-18:
The SMA integration was added by @cjthuys , maybe you could take a look?

pb66, 2018-01-18:
Although I would like to see a fix in emonhub too, I think emoncms should handle "NaN"s and "null"s too, I have noticed in the past that a simple divide by zero in a sketch running on a serially connected emontx passes a NaN, which emonhub has passed faithfully (rightly or wrongly) to emoncms and because emoncms just rejects the who;e request and emonhub keeps trying to redeliver the failed payload, no future data is passed successfully until that bad value is removed from the buffer.

Since emoncms can handle nulls eg [(123,456,null,789)] is valid, the 4th input simply isn't updated, I believe the same should happen for [(123,456,NaN,789)] or for that matter any other non-numeric value.

this isn't so much about the corrupt data being ignored but more about the good data in the same payload being ignored because of one bad value.