openenergymonitor/emonhub #48: Dead link: EmonHub config file -> Nodes

Deathlesss, 2018-02-02:
The link in EmonHub config file under Node is 404:

`[nodes]

## See config user guide: http://github.com/openenergymonitor/emonhub/blob/master/configuration.md
`

I assume it has to go to: https://github.com/openenergymonitor/emonhub/blob/emon-pi/configuration.md

??

Deathlesss, 2018-02-02:
HM... on second thoughts, I guess the configuration does NOT update when an new version comes. So I have to manually update this text from [here](https://github.com/openenergymonitor/emonhub/blob/emon-pi/conf/emonpi.default.emonhub.conf) right?

https://github.com/openenergymonitor/emonhub/blob/emon-pi/conf/emonpi.default.emonhub.conf

glynhudson, 2018-02-02:
Thanks, Fixed 👍

closed 2018-02-02 by glynhudson

Deathlesss, 2018-02-02:
thanks for fixing!

commit 56a65ee3 (2018-02-02): URL corrections fixes #48