openenergymonitor/emonhub #50: Modbus interfacer in v2.1.0

TrystanLea, 2018-02-19:
Hello @cjthuys,  I had a bit of trouble re-factoring the modbus interfacer to work with my recent update to emonhub where I've attempted to clean up the interfacer loading code in emonhub.py. Ideally all checking for the existence of imported modules that are specific to an interfacer should be done inside that interfacer. I also dont have a fronius modbus inverter to test with. Would you mind taking a look at the interfacer? I've put it in a folder called tmp temporarily https://github.com/openenergymonitor/emonhub/blob/emon-pi/src/interfacers/tmp/EmonFroniusModbusTcpInterfacer.py

alexandrecuer, 2018-03-04:
Hello
I am not cthuys but I am using a lot the modbus interfacer....I am reinstalling a couple of emonCMS RPI server on some of the sites I am monitoring....I am mostly working with industrial modbus IP masters (just like sofrel S500 wich is intended to manage large energy/water installations)  
I will try to test the new interfacer in the month....
Regards
Alex

alexandrecuer, 2018-03-05:
@TrystanLea I've tried to use version 2.1.1 of emonhub with the modbusTCPinterfacer....I've got the following error : "MainThread Unable to create 'ModbusTCP' interfacer: global name 'pymodbus_found' is not defined"
maybe it is due to a wrong config of pymodbus
I've got python version 2.7.9 and pymodbus v1.4.0 but without twisted...

alexandrecuer, 2018-03-05:
I commented line 109 and 110 in emonhub.conf

```
#if I['Type'] in ('EmonModbusTcpInterfacer','EmonFroniusModbusTcpInterfacer') and not pymodbus_found :
#self._log.error("Python module pymodbus not installed. unable to load modbus interfacer")
```

the ModbusTcpinterfacer starts but the following errors are returned

```
2018-03-05 11:29:08,440 WARNING  ModbusTCP  Exception caught in ModbusTCP thread. Traceback (most recent call last):
File "/home/pi/emonhub/src/emonhub_interfacer.py", line 40, in wrapper
return f(*args)
File "/home/pi/emonhub/src/emonhub_interfacer.py", line 102, in run
rxc = self.read()
File "/home/pi/emonhub/src/interfacers/EmonModbusTcpInterfacer.py", line 108, in read
decoder = BinaryPayloadDecoder.fromRegisters(self.rVal.registers, endian=Endian.Big)
TypeError: fromRegisters() got an unexpected keyword argument 'endian'
```

TrystanLea, 2018-03-05:
Hello @alexandrecuer version 2.1.0 temporarily disables support for the modbus interfacer as I couldn’t get it to work either. I was hoping @cjthuys might be able to take a look at it. Did you move the interfacer out of the tmp folder to try it?

alexandrecuer, 2018-03-05:
Yes I dit move the EmonModbusTcpInterfacer.py out of tmp folder

alexandrecuer, 2018-03-05:
OK I manage to make it work.....the function BinaryPayloadDecoder.fromRegisters has changed in pymodbus 1.4.0
in EmonModbusTcpInterfacer.py, line 108, it must be :
  `decoder = BinaryPayloadDecoder.fromRegisters(self.rVal.registers, byteorder=Endian.Big, wordorder=Endian.Big)`
I can submit a pull request if you wish...
anyway, this is only related to the main EmonModbusTcpInterfacer.py file, not to the ones dedicated to specific equipments (Fronius and al....)

TrystanLea, 2018-03-05:
Thanks @alexandrecuer a pull request would be much appreciated!

alexandrecuer, 2018-03-05:
Just done
assures compatibility with pymodbus 1.4.0

closed 2020-01-30 by TrystanLea