openenergymonitor/emonhub #65: EmonHubGraphiteInterfacer.py is not submitting data

robbrad, 2018-07-02:
When using this interfacer Im getting the following error

2018-07-01 22:24:07,148 WARNING  Graphite   Exception caught in Graphite thread. Traceback (most recent call last):
  File "/home/pi/emonhub/src/emonhub_interfacer.py", line 40, in wrapper
return f(*args)
  File "/home/pi/emonhub/src/emonhub_interfacer.py", line 128, in run
self.action()
  File "/home/pi/emonhub/src/emonhub_interfacer.py", line 198, in action
self.flush()
  File "/home/pi/emonhub/src/emonhub_interfacer.py", line 216, in flush
if self._process_post(databuffer):
  File "/home/pi/emonhub/src/interfacers/EmonHubGraphiteInterfacer.py", line 69, in _process_post
timestamp = frame['timestamp']
KeyError: 'timestamp'

This is because a timestamp value is not set

Changing the code to the below code fixes that

 def _process_post(self, databuffer):

        timestamp = int(time.time())

        metrics = []

        for c in range(0,len(databuffer)):
            frame = databuffer[c]
            nodename = frame['node']

            for inputname,value in frame['data'].iteritems():
                # path
                path = self._settings['prefix']+'.'+nodename+"."+inputname
                # payload
                payload = str(value)
                # timestamp
                # timestamp = frame['timestamp']
                metrics.append(path+" "+payload+" "+str(timestamp))
                #metrics.append(path+" "+payload+" 1234")

duckman, 2019-04-24:
I just ran into this today. Code change got things working for me as well.

glynhudson, 2019-04-24:
Sorry I can't what's different between your code and the current code: https://github.com/openenergymonitor/emonhub/blob/emon-pi/src/interfacers/EmonHubGraphiteInterfacer.py#L56

Please could you submit this as a pull request. If you not familer with git the easiest way for a simple change is via the online editor: https://help.github.com/en/articles/editing-files-in-another-users-repository

duckman, 2019-04-25:
#79

closed 2020-01-30 by TrystanLea