openenergymonitor/emonhub #73: Dial / Feedvalue / Jguage / Jguage2 - now missing

fritserasmus, 2019-01-15:
I used to have a couple of dashboards reporting PV power produced / Power used in house and surplus fed back into the grid. 

So I wanted to make sure all is up to date in the administration Menu option and decided it is probably good to run an update on "emonPi" - 

Gone are all my nice dials I used to have.

No matter what I tried after that, I am not able to get them to show up again. Am I just doing something wrong, OR was the dials etc removed in recent update 9.9.3 (The one I updated to)

Below is an example of such a dashboard reading my Axpert inverter and not the emonCMS, illustrating some of the info I would love to have back:

[image]

Below is a screenshot of the Toolbox how it used to be before; now the options are not there any more:

[image]

Can anyone perhaps give me some pointers how to get the dials and real time values back on the dashboards?

Thanking in advance for the help!!
Frits

pb66, 2019-01-15:
Hi @fritserasmus this is an emoncms issue unrelated to emonhub, please open an issue at either [emoncms/emoncms](https://github.com/emoncms/emoncms) or more likely [emoncms/dashboard](https://github.com/emoncms/dashboard)? However you might be better off starting with a post on the [OEM forum](https://community.openenergymonitor.org/) first as that is a more central point and gets more traffic (helpers).

pb66, 2019-01-15:
OEM forum discussion started at https://community.openenergymonitor.org/t/dashboards-widgets-missing-since-emonpi-update/9770?u=pb66. 

[This issue can now be closed]

closed 2020-01-29 by TrystanLea