openenergymonitor/emonhub #78: Install script

borpin, 2019-04-05:
https://github.com/openenergymonitor/emonhub/blob/cb9da3c14babf8d7f0c2203450465b6e9111c99b/install.sh#L19

If I wanted to install just the emonhub onto a raw Raspbian, to take serial data and send it to emoncms, would this do it?

Are there assumptions re the Python Environment (2.7 or 3.x) that pip is installed, that git is installed?

Just asking for a friend 😄 

Can the script be named `install-emonbub.sh`?

beaylott, 2019-04-10:
It maybe an idea to move to a virtual environment which can be setup from a requirements.txt file distributed with the repository. This is fairly standard practice these days for Python. 

And I maybe wrong but I think emonhub is on 2.7 still.

borpin, 2020-06-09:
Yes is the short answer.

closed 2020-06-09 by borpin