openenergymonitor/emonhub #80: Data corrupted between MQTT and Jee Interfacers

delboy711, 2019-04-28:
I am sending transmit data via MQTT to a remote node. The data is a single value sent to Node 24.
The Jee interfacer passes the message apparently OK, but it gets transmitted to the wrong node ID, and the message is sent as 3 bytes instead of 2.

Log sequence is 
MQTT       Nodeid: 24 values: 1526
MQTT       9 Sent to channel' : ToRFM12
RFM2Pi     9 sent TX packet: 24,246,5,s
acknowledged command: > 24,246,5,0s
confirmed sent packet size: -> 3 b

The packet arrives at Node 5

The stanza in emonhub.conf for the node is

[[24]]
    nodename = solar_EVSE
    firmware =solar_EVSE
    hardware = solar_EVSE
    [[[tx]]]
        names = dacsetpoint
        units = C
        scales = 1
datacode = h

I can get it working by patching EmonHubJeeInterfacer.py  send() function
`    def send (self, cargo):

        f = cargo
        cmd = "s"

        if self.getName() in f.encoded:
            data = f.encoded[self.getName()]
        else:
            data = f.realdata

        payload = ""
        for value in range(1, len(data)):
            if int(data[value]) < 0 or int(data[value]) > 255:
                self._log.warning(self.name + " discarding Tx packet: values out of scope" )
                return
            payload += str(int(data[value]))+","

        payload += data[0] + cmd

        self._log.debug(str(f.uri) + " sent TX packet: " + payload)
        self._ser.write(payload)
`

The log sequence then looks like
MQTT       Nodeid: 24 values: 1526
MQTT       23 Sent to channel' : ToRFM12
 RFM2Pi     23 sent TX packet: 246,5,24s
RFM2Pi     acknowledged command: > 246,5,24s
RFM2Pi     confirmed sent packet size: -> 2 b

The node receives that OK

It is not clear to me if it is the MQTT Interfacer malforming the command, or the Jee Interfacer misreading it. If the latter, then here is the fix:-)

Thanks

Derek

delboy711, 2022-06-27:
This is a blast from the past! Thanks for getting to the bottom of this one Alan.
Now I need to hunt out the source for my application.

alandpearson, 2022-08-12:
Fixed in main codebase, pull request submitted : 
https://github.com/openenergymonitor/emonhub/pull/175