openenergymonitor/emonhub #90: python3 - serial interfacer

borpin, 2020-03-24:
@bwduncan

Using either the `EmonHubSerialInterfacer` or the `EmonHubTx3eInterfacer` the following error is seen;

```
2020-03-24 14:43:08,760 WARNING  MainThread Attempting to restart thread SerialTx (thread has been restarted 68 times...
2020-03-24 14:43:08,763 INFO     MainThread Creating EmonHubSerialInterfacer 'SerialTx'
2020-03-24 14:43:08,769 DEBUG    MainThread Opening serial port: /dev/ttyAMA0 @ 115200 bits/s
2020-03-24 14:43:08,772 DEBUG    MainThread Setting SerialTx pubchannels: ['ToEmonCMS']
2020-03-24 14:43:18,742 WARNING  SerialTx   Exception caught in SerialTx thread. Traceback (most recent call last):
  File "/opt/openenergymonitor/emonhub/src/emonhub_interfacer.py", line 32, in wrapper
    return f(*args)
  File "/opt/openenergymonitor/emonhub/src/emonhub_interfacer.py", line 99, in run
    rxc = self.read()
  File "/opt/openenergymonitor/emonhub/src/interfacers/EmonHubSerialInterfacer.py", line 90, in read
    c.nodeid = int(f[0])
ValueError: invalid literal for int() with base 10: 'MSG:71923,Vrms:246.78,P1:157,E1:16346,pulse:15'

2020-03-24 14:43:18,758 WARNING  MainThread SerialTx thread is dead.
```

This is a shop emonTX - no mods.

bwduncan, 2020-03-24:
Hey I've been looking at the code and I'm not sure how it could work with EmonHubSerialInterfacer. Can you show the error you get with EmonHubTx3eInterfacer?

borpin, 2020-03-26:
Sorry, missed this comment. the error is exactly the same.

The difference between the 2 is that the string it expects is different (I believe).

bwduncan, 2020-03-26:
I would expect the Tx3 interface to handle that string without any problems, so if the exception is the same then something weird is going on. Maybe it's running the wrong code?

borpin, 2020-03-26:
When I switch to the emon-pi branch and restart, it is fine.

borpin, 2020-03-26:
So checking again,

error message
```
2020-03-26 12:42:06,468 WARNING  MainThread Attempting to restart thread SerialTx (thread has been restarted 220 times...
2020-03-26 12:42:06,473 INFO     MainThread Creating EmonHubTx3eInterfacer 'SerialTx'
2020-03-26 12:42:06,481 DEBUG    MainThread Opening serial port: /dev/ttyAMA0 @ 115200 bits/s
2020-03-26 12:42:06,490 WARNING  SerialTx   Exception caught in SerialTx thread. Traceback (most recent call last):
  File "/opt/openenergymonitor/emonhub/src/emonhub_interfacer.py", line 32, in wrapper
    return f(*args)
  File "/opt/openenergymonitor/emonhub/src/emonhub_interfacer.py", line 99, in run
    rxc = self.read()
  File "/opt/openenergymonitor/emonhub/src/interfacers/EmonHubTx3eInterfacer.py", line 46, in read
    self._rx_buf = self._rx_buf + self._ser.readline()
TypeError: can only concatenate str (not "bytes") to str
```

As there is nothing in the file header, version `#!` or anything, it is difficult to ensure it is the right file.

Should there be a python3 #!

bwduncan, 2020-03-26:
That's more like it! Pop a `.decode()` on the end of /opt/openenergymonitor/emonhub/src/interfacers/EmonHubTx3eInterfacer.py line 46

borpin, 2020-03-26:
literally like this?
[image]

borpin, 2020-03-26:
Magic - solved.

borpin, 2020-03-26:
Will you do a PR or shall I?

bwduncan, 2020-03-26:
I've added it to my branch and I'll make a PR soon. It looks like that one just got missed by accident. All the other serial interfacers already have the correct text handling. Thanks!

borpin, 2020-05-01:
Tripped up on this again 😭 on a fresh install.

borpin, 2020-06-09:
Fixed.

closed 2020-06-09 by borpin