openenergymonitor/emonhub #125: HTTP interfacer stops instead of retrying

borpin, 2020-08-20:
https://community.openenergymonitor.org/t/emonhub-stops-sending-data-to-emoncms-org/14639/43

Another report of the HTTP interfacer stopping when it encounters a network/server issue. It should buffer the data and keep trying.

This has been introduced with the Python3 work I believe @bwduncan @TrystanLea.

bwduncan, 2020-08-20:
I can't reproduce this bug. When I firewall off emoncms.org, emonhub (or rather, requests) tries really, really hard to send the data. It gives up after about 2 minutes and reports a "send failure" warning log, but the data stays in the buffer. When I remove the firewall block, it sends the buffered data and carries on happily.

```
2020-08-20 12:43:49,779 DEBUG    emoncmsorg Buffer size: 7
2020-08-20 12:43:49,781 INFO     emoncmsorg sending: https://emoncms.org/input/bulk.json?apikey=E-M-O-N-C-M-S-A-P-I-K-E-Y&data=[[1597927402.280812,5,-1097,1529,432,238.14000000000001,18.1,0,0,0,0,0,2],[1597927407.326337,5,-1080,1541,461,238.62,18,0,0,0,0,0,2],[1597927409.6592908,23,20.8,22.1,79.5,2.4000000000000004,1,-56],[1597927412.3111854,5,-999,1549,550,238.27,18,0,0,0,0,0,2],[1597927417.2784438,5,-961,1451,490,238.1,18,0,0,0,0,0,2],[1597927422.325498,5,-297,788,491,237.9,18,0,0,0,0,0,2],[1597927427.2861547,5,-341,879,538,237.63,18.1,0,0,0,0,0,2]]&sentat=1597927429
2020-08-20 12:45:59,058 WARNING  emoncmsorg emoncmsorg couldn't send to server: HTTPSConnectionPool(host='emoncms.org', port=443): Max retries exceeded with url: /input/bulk.json?apikey=1931ca22b4a6a8d4dfe00c130573b21c (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x74b84450>: Failed to establish a new connection: [Errno 110] Connection timed out'))
2020-08-20 12:45:59,059 WARNING  emoncmsorg send failure: wanted 'ok' but got 'None'
2020-08-20 12:46:29,118 DEBUG    emoncmsorg Buffer size: 42
2020-08-20 12:46:29,122 INFO     emoncmsorg sending: https://emoncms.org/input/bulk.json?apikey=E-M-O-N-C-M-S-A-P-I-K-E-Y&data=[[1597927402.280812,5,-1097,1529,432,238.14000000000001,18.1,0,0,0,0,0,2],[1597927407.326337,5,-1080,1541,461,238.62,18,0,0,0,0,0,2],[1597927409.6592908,23,20.8,22.1,79.5,2.4000000000000004,1,-56],[1597927412.3111854,5,-999,1549,550,238.27,18,0,0,0,0,0,2],[1597927417.2784438,5,-961,1451,490,238.1,18,0,0,0,0,0,2],[1597927422.325498,5,-297,788,491,237.9,18,0,0,0,0,0,2],[1597927427.2861547,5,-341,879,538,237.63,18.1,0,0,0,0,0,2],[1597927432.3329237,5,-935,1491,556,237.71,18,0,0,0,0,0,2],[1597927437.2890184,5,-497,1043,546,237.69,18,0,0,0,0,0,2],[1597927442.3426878,5,-439,965,526,238.01,18,0,0,0,0,0,2],[1597927447.2924068,5,-645,1219,574,238.3,18,0,0,0,0,0,2],[1597927452.3413112,5,-882,1455,573,238.27,18.1,0,0,0,0,0,2],[1597927457.2926416,5,-752,1369,617,237.74,18,0,0,0,0,0,2],[1597927462.3395386,5,-849,1483,634,238.31,18.1,0,0,0,0,0,2],[1597927467.304761,5,-787,1443,656,238.74,18,0,0,0,0,0,2],[1597927469.3287232,23,20.8,22.1,79,2.4000000000000004,1,-56],[1597927472.3620732,5,-859,1545,686,238.98000000000002,18.1,0,0,0,0,0,2],[1597927477.3369164,5,-860,1539,679,238.61,18.1,0,0,0,0,0,2],[1597927482.3775783,5,-825,1544,719,238.63,18,0,0,0,0,0,2],[1597927487.3815553,5,-1042,1550,508,239.17000000000002,18.1,0,0,0,0,0,2],[1597927492.332222,5,-1107,1553,446,238.95000000000002,18,0,0,0,0,0,2],[1597927497.3036237,5,-499,882,383,238.21,18.1,0,0,0,0,0,2],[1597927502.3638732,5,-96,537,441,237.91,18.1,0,0,0,0,0,2],[1597927507.3330371,5,-12,448,436,238.18,18.1,0,0,0,0,0,2],[1597927512.3018909,5,50,408,458,237.68,18.1,0,0,0,0,0,2],[1597927517.3606389,5,49,388,437,237.70000000000002,18.1,0,0,0,0,0,2],[1597927522.316835,5,-7,445,438,237.8,18.1,0,0,0,0,0,2],[1597927527.3683789,5,-51,474,423,237.68,18.1,0,0,0,0,0,2],[1597927528.9971104,23,20.8,22.1,78.60000000000001,2.4000000000000004,1,-56],[1597927532.339185,5,47,405,452,238.46,18,0,0,0,0,0,2],[1597927537.3953216,5,38,406,444,238.38,18.1,0,0,0,0,0,2],[1597927542.3539228,5,41,416,457,237.6,18.1,0,0,0,0,0,2],[1597927547.3036234,5,19,429,448,238.03,18.1,0,0,0,0,0,2],[1597927552.3618088,5,-18,426,408,237.44,18.1,0,0,0,0,0,2],[1597927557.310831,5,-12,424,412,238.20000000000002,18.1,0,0,0,0,0,2],[1597927562.3593416,5,-28,441,413,238.33,18.1,0,0,0,0,0,2],[1597927567.3343713,5,-99,487,388,238.5,18.1,0,0,0,0,0,2],[1597927572.3881333,5,-81,477,396,238.27,18.1,0,0,0,0,0,2],[1597927577.3364644,5,-25,513,488,237.99,18.1,0,0,0,0,0,2],[1597927582.3832214,5,-65,516,451,238.53,18.1,0,0,0,0,0,2],[1597927587.3390226,5,-106,604,498,237.70000000000002,18.1,0,0,0,0,0,2],[1597927588.5777912,23,20.700000000000003,22.1,78.4,2.4000000000000004,1,-57]]&sentat=1597927589
2020-08-20 12:47:01,396 DEBUG    emoncmsorg acknowledged receipt with 'ok' from https://emoncms.org
```

borpin, 2020-08-20:
Is it the type of error received that is the problem?

I note that the last log message received by the user is a `sending`, so it is not then logging a `send failure` or an `acknowledged`  response.

It isn't an isolated incident so something is off somewhere.

borpin, 2020-08-20:
What will an exception here do

https://github.com/openenergymonitor/emonhub/blob/6820033f67b556dab5f67b4f400be13bf18a1476/src/interfacers/EmonHubEmoncmsHTTPInterfacer.py#L71-L71

Should it raise an exception ever?

bwduncan, 2020-08-20:
Given that requests is trying so hard (at least 2 minutes for a quick failure, could be much longer if the server responds slowly, or starts responding and then gives up) it could be that we're just not waiting long enough.

If there was an exception not being caught (and as you point out, all requests exceptions inherit from `RequestException`, it would be logged by the emonhub main loop, the thread would be killed and restarted. I'm not seeing any evidence of this happening. If it were, the buffered data would be thrown away and fresh data would be sent.

In any case, unless there is a success or error message being logged, requests is still trying.

bwduncan, 2020-08-20:
Can you reproduce this? Here are the commands I used to block emoncms.org:

```
# insert a drop rule for emoncms.org
sudo iptables -I OUTPUT --destination emoncms.org -j DROP
# delete the drop rule
sudo iptables -D OUTPUT --destination emoncms.org -j DROP
```

borpin, 2020-08-20:
> If there was an exception not being caught (and as you point out, all requests exceptions inherit from `RequestException`, it would be logged by the emonhub main loop,

Well maybe. I have seen exceptions *not* appearing in the emonhub log and only in syslog when I was playing with that pulse interfacer.

Could there be an exception raised that is not part of `RequestException` at this point?

> In any case, unless there is a success or error message being logged, requests is still trying.
```
2020-08-20 15:51:43,688 INFO     emoncmsorg sending:
```
It seems this message is always the last one logged (see other reports on forum). How can that be?

Bit of a Sherlock moment; When you have eliminated the impossible, what remains however improbable must be the truth.

> Can you reproduce this? Here are the commands I used to block emoncms.org:

I don't think it is a complete block that is the issue. I think it is a different error/exception that is causing the problem.

If you remove the handler and block emoncms.org, how does the script fail? do you get a `sending` as the last log message?

bwduncan, 2020-08-20:
> Well maybe. I have seen exceptions _not_ appearing in the emonhub log and only in syslog when I was playing with that pulse interfacer.
>
> Could there be an exception raised that is not part of [...]

It's not impossible, but I'm not aware of any, and `log_exceptions_from_class_method` (at the top of emonhub_interfacer.py) will catch *all* exceptions and log them.

> > In any case, unless there is a success or error message being logged, requests is still trying.
> 
> ```
> 2020-08-20 15:51:43,688 INFO     emoncmsorg sending:
> ```
> 
> It seems this message is always the last one logged (see other reports on forum). How can that be?

Not waiting long enough! Once `requests.post` returns/raises there are only 4 code paths: log success, log error, log `RequestException` (in `_send_post`) or log another exception (in `log_exceptions_from_class_method`). Anything else (except for emonhub dying violently via an uncatchable signal like `SIGKILL` or the power being pulled) is impossible.

> Bit of a Sherlock moment; When you have eliminated the impossible, what remains however improbable must be the truth.

I thought that was Spock?

> > Can you reproduce this? Here are the commands I used to block emoncms.org:
> 
> I don't think it is a complete block that is the issue. I think it is a different error/exception that is causing the problem.

There are other ways we could introduce errors. You could point emonhub at a server which will just return 404, or you could set up a proxy which sends one byte every minute.

> If you remove the handler and block emoncms.org, how does the script fail? do you get a `sending` as the last log message?

Which handler? The one in `_send_post`? If I remove that and then block emoncms.org in the firewall, I have to wait *a really long time* and then I eventually get:

```
2020-08-20 17:08:07,740 WARNING  emoncmsorg Exception caught in emoncmsorg thread. Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 160, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/usr/local/lib/python3.7/dist-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/usr/local/lib/python3.7/dist-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/urllib3/connectionpool.py", line 603, in urlopen
    chunked=chunked)
  File "/usr/local/lib/python3.7/dist-packages/urllib3/connectionpool.py", line 344, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.7/dist-packages/urllib3/connectionpool.py", line 843, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 316, in connect
    conn = self._new_conn()
  File "/usr/local/lib/python3.7/dist-packages/urllib3/connection.py", line 169, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x75cda2d0>: Failed to establish a new connection: [Errno 110] Connection tim
ed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/local/lib/python3.7/dist-packages/urllib3/connectionpool.py", line 641, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/local/lib/python3.7/dist-packages/urllib3/util/retry.py", line 399, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='emoncms.org', port=443): Max retries exceeded with url: /input/bulk.json?apikey=1931ca22b4a6a8d4dfe00c130573b21c (Caused by NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x75cda2d0>: Failed to establish a new connection: [Errno 110] Connection timed out'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... 41 lines cut ...]
2020-08-20 17:08:08,776 INFO     MainThread Creating EmonHubEmoncmsHTTPInterfacer 'emoncmsorg' 
2020-08-20 17:08:08,778 DEBUG    MainThread Setting emoncmsorg pubchannels: ['ToRFM12']
2020-08-20 17:08:08,779 DEBUG    MainThread Setting emoncmsorg subchannels: ['ToEmonCMS']
2020-08-20 17:08:08,780 INFO     MainThread Setting emoncmsorg apikey: set
2020-08-20 17:08:08,780 INFO     MainThread Setting emoncmsorg url: https://emoncms.org
2020-08-20 17:08:08,781 INFO     MainThread Setting emoncmsorg senddata: 1
2020-08-20 17:08:08,782 INFO     MainThread Setting emoncmsorg sendstatus: 1
2020-08-20 17:08:09,128 DEBUG    MQTT       Publishing: emon/emonpi/power1 229
2020-08-20 17:08:09,130 DEBUG    MQTT       Publishing: emon/emonpi/power2 144
2020-08-20 17:08:09,132 DEBUG    MQTT       Publishing: emon/emonpi/power1pluspower2 373
2020-08-20 17:08:09,134 DEBUG    MQTT       Publishing: emon/emonpi/vrms 238.76
2020-08-20 17:08:09,135 DEBUG    MQTT       Publishing: emon/emonpi/t1 18.5
2020-08-20 17:08:09,137 DEBUG    MQTT       Publishing: emon/emonpi/t2 0
2020-08-20 17:08:09,139 DEBUG    MQTT       Publishing: emon/emonpi/t3 0
[... cut ...]
2020-08-20 17:08:09,185 DEBUG    emoncmsorg Buffer size: 1
2020-08-20 17:08:09,186 INFO     emoncmsorg sending: https://emoncms.org/input/bulk.json?apikey=E-M-O-N-C-M-S-A-P-I-K-E-Y&data=[[1597943288.1168897,5,229,144,373,238.76,18.5,0,0,0,0,0,2]]&sentat=1597943289
```

`requests.exceptions.ConnectionError` of course being a subclass of `requests.exceptions.RequstError` which we would have previously caught and retained the buffer contents.

But "sending" is not the "last" message... just the most recent one for a while!

borpin, 2020-08-20:
> But "sending" is not the "last" message... just the most recent one for a while!

Ok, but users report it stopping for hours, so something isn't right! They also report it never recovers. In any case, I would expect it to timeout long before that. I seem to remember one report where it had stopped for over a day.

> I have to wait _a really long time_ and then I eventually get:

But not hours.

> I thought that was Spock?

A bit before his time (1890) https://en.wikiquote.org/wiki/Sherlock_Holmes

What else could it be?  The fact that just restarting emonhub restores communications indicates that is the source.

bwduncan, 2020-08-20:
> Ok, but users report it stopping for hours, so something isn't right! They also report it never recovers. In any case, I would expect it to timeout long before that. I seem to remember one report [...]

Yes it seems there is something not right.

Incidentally, requests will happily hang forever if it starts receiving a response and then the server just quietly goes away. I will add a timeout to the requests calls (which is strongly recommended in the documentation)

> > I have to wait _a really long time_ and then I eventually get:
> 
> But not hours.

No, a few minutes.

> > I thought that was Spock?
> 
> A bit before his time (1890) https://en.wikiquote.org/wiki/Sherlock_Holmes
> 
> What else could it be? The fact that just restarting emonhub restores communications indicates that is the source.

It's hard to say without being able to reproduce it. I'll see if I can get a tar pit firewall rule to make it go slowly, but an indefinitely hanging socket seems likely.

Whatever is happening, it looks like `requests.post` never returns or raises an exception...

> [edit]

Please can you stop editing comments? It makes it hard to keep up when reading the emails!

> Looked back through the thread, and yes a user reports 48 hours
> 
> https://community.openenergymonitor.org/t/emonhub-stops-sending-data-to-emoncms-org/14639/28?u=borpin
> 
> I think we need to add some additional logging in to try and determine what is happening.

Looking at that user's post, it looks like they have logging set to WARNING, rather than DEBUG. Turning up the debugging may help.

Bruce

borpin, 2020-08-20:
> Incidentally, requests will happily hang forever if it starts receiving a response and then the server just quietly goes away. I will add a timeout to the requests calls (which is strongly recommended in the documentation)

Sounds like a good plan. Will it generate a specific timeout log message? I suggest that should be a WARN.

bwduncan, 2020-08-20:
https://github.com/openenergymonitor/emonhub/pull/126

Yes, it will generate a WARNING message with "couldn't send to server".

Bruce

borpin, 2020-08-20:
Could it be more specific that it timed out?

bwduncan, 2020-08-20:
I think the exception message will be descriptive enough:

```
In [5]: try: 
   ...:     requests.get('http://google.com', timeout=0.0001) 
   ...: except requests.exceptions.RequestException as ex: 
   ...:     print("couldn't send to server: " + str(ex)) 
   ...:                                                                                                                                                                   
couldn't send to server: HTTPConnectionPool(host='google.com', port=80): Max retries exceeded with url: / (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f1e4bde1be0>, 'Connection to google.com timed out. (connect timeout=0.0001)'))
```

borpin, 2020-08-23:
> Aug 21 23:35:32 emonpi dhcpcd[502]: wlan0: carrier lost

I have seen this happen where the Country code on the Router is different to the Country Code on the wlan interface.

There was a discussion somewhere about it (either community or GitHub).

borpin, 2020-08-24:
Another report of this but not just on an emonPi setup. @TrystanLea 

https://community.openenergymonitor.org/t/emonhub-stops-sending-data-to-emoncms-org/14639/45?u=borpin

bwduncan, 2020-08-24:
> In my opinion it is unlikely to reproduce the issue by blocking emoncms.org.
> I think the issue appears when the wireless interface flaps.

I'm not sure what the difference is as far as the code is concerned. If the wireless interface goes down, no more packets will be received. That's the same as having the firewall drop the packets.

> Here is a relevant portion of the emonhub.log:
> 
> ```
> 2020-08-21 23:35:06,780 INFO     MQTT       Publishing: emonhub/rx/5/values 268,0,268,230.88,25.6,0,0,0,0,0,0
> 2020-08-21 23:35:10,477 DEBUG    emoncmsorg Buffer size: 7
> 2020-08-21 23:35:10,479 INFO     emoncmsorg sending: https://emoncms.org/input/bulk.json?apikey=E-M-O-N-C-M-S-A-P-I
> -K-E-Y&data=[[1598042081.5493429,5,270,0,270,230.4,25.6,0,0,0,0,0,0],[1598042086.498806,5,260,0,260,228.76,25.6,0,0
> ,0,0,0,0],[1598042091.4396791,5,268,0,268,227.94,25.6,0,0,0,0,0,0],[1598042096.4766777,5,262,0,262,228.55,25.6,0,0,
> 0,0,0,0],[1598042101.5275505,5,262,0,262,230.65,25.6,0,0,0,0,0,0],[1598042102.1392407,24,27.400000000000002,0,48.1,
> 2.5,1,-69],[1598042106.469771,5,268,0,268,230.88,25.6,0,0,0,0,0,0]]&sentat=1598042110
> 2020-08-21 23:35:11,516 DEBUG    RFM2Pi     64723 NEW FRAME : OK 5 22 1 0 0 22 1 222 89 0 1 0 0 0 0 0 0 0 0 0 0 0 0
>  0 0 (-0)
> ```
> 
> After this, it never tries to send again (I left it for 12 hours this time).

Can you gzip the entire log and post it here?

> and this is syslog:
> 
> ```
> Aug 21 23:34:35 emonpi dhcpcd[502]: wlan0: deleting route to 169.254.0.0/16
> Aug 21 23:34:35 emonpi dhcpcd[502]: wlan0: pid 502 deleted default route via 172.17.1.1
> Aug 21 23:35:01 emonpi CRON[20426]: (root) CMD (/usr/local/bin/wifi-check > /var/log/emoncms/wificheck.log 2>&1)
> Aug 21 23:35:32 emonpi dhcpcd[502]: wlan0: carrier lost
> Aug 21 23:35:32 emonpi dhcpcd[502]: wlan0: deleting address fe80::c832:2e4c:8ac1:d796
> Aug 21 23:35:32 emonpi avahi-daemon[371]: Withdrawing address record for fe80::c832:2e4c:8ac1:d796 on wlan0.
> Aug 21 23:35:32 emonpi avahi-daemon[371]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::c
> 832:2e4c:8ac1:d796.
> Aug 21 23:35:32 emonpi avahi-daemon[371]: Interface wlan0.IPv6 no longer relevant for mDNS.
> Aug 21 23:35:32 emonpi avahi-daemon[371]: Withdrawing address record for 172.17.1.106 on wlan0.
> Aug 21 23:35:32 emonpi avahi-daemon[371]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 172.17.
> 1.106.
> Aug 21 23:35:32 emonpi avahi-daemon[371]: Interface wlan0.IPv4 no longer relevant for mDNS.
> Aug 21 23:35:32 emonpi dhcpcd[502]: wlan0: deleting route to 172.17.1.0/24
> Aug 21 23:35:32 emonpi dhcpcd[502]: wlan0: carrier acquired
> Aug 21 23:35:32 emonpi dhcpcd[502]: wlan0: IAID eb:c6:c5:69
> ```

Same.
 
> If you guys need more info, I am happy to help - I see the issue happening once every 2-3 days.

I would like you to test two things:

1. Find a way to reproduce the bug. If we don't have a test then we can never know if it's fixed. If you think it's the wireless interface then just pull the dongle.

2. Having done that, test the PR that I posted above (#126)

Thanks,
Bruce

lgheorghe, 2020-08-24:
I would post the logs, but I'm afraid I can't - I copied and gunzipped in the /tmp directory and we just had a power outage today. I looked in the /var/log.old directory and I think they where rotated.
I will have to wait for the next time it happens.

Regarding the tests:

1. I'm using the onboard wireless interface - no dongle to pull. I was thinking of using the wired ethernet interface, but I need a quite log cable to do that.

2. I need some more details on how to test PR #126 

Thanks,
Liviu

bwduncan, 2020-08-24:
That's frustrating.

1. Maybe you could bring the interface down in software? `ip link set wlan0 down` or something
2. If you're not comfortable using git then it would be easy to get the installed software into a mess. Let's look into that once we have a reproducible test.

Bruce

borpin, 2020-08-26:
@bwduncan could the problem be out of date (or different from you) Python packages?

@lgheorghe, could you do
```
pip3 list
```
and post the output.

I discovered that quite a few of my Python packages are not at the latest version.

lgheorghe, 2020-08-26:
@borpin, the requested output
```
pi@emonpi:~ $ pip3 list
Package          Version  
---------------- ---------
asn1crypto       0.24.0   
attrs            18.2.0   
Automat          0.6.0    
certifi          2018.8.24
chardet          3.0.4    
Click            7.0      
colorama         0.3.7    
colorzero        1.1      
configobj        5.0.6    
constantly       15.1.0   
cryptography     2.6.1    
entrypoints      0.3      
gpiozero         1.5.1    
hyperlink        17.3.1   
idna             2.6      
incremental      16.10.1  
keyring          17.1.1   
keyrings.alt     3.1.1    
paho-mqtt        1.5.0    
pip              18.1     
pyasn1           0.4.2    
pyasn1-modules   0.2.1    
pycrypto         2.6.1    
PyGObject        3.30.4   
pymodbus         2.1.0    
pyOpenSSL        19.0.0   
pyserial         3.4      
pyserial-asyncio 0.4      
python-apt       1.8.4.1  
pyxdg            0.25     
redis            3.5.3    
requests         2.21.0   
RPi.GPIO         0.7.0    
SecretStorage    2.3.1    
service-identity 16.0.0   
setuptools       40.8.0   
six              1.12.0   
spidev           3.4      
ssh-import-id    5.7      
Twisted          18.9.0   
ufw              0.36     
urllib3          1.24.1   
wheel            0.32.3   
xmltodict        0.12.0   
zope.interface   4.3.2  
```

borpin, 2020-08-26:
@bwduncan - could you compare this against what you are running? For me the `urllib3` package seems to be upgradable.

bwduncan, 2020-09-01:
OK, we're getting deep into a rabbit hole here without really knowing why, but why not, it sounds like fun.

pip is going to be listing all the installed packages it can find, not just the user installed ones. There is an important difference between the system-installed packages (we shouldn't mess with those) and the user installed ones (we might be responsible for those). For starters, we need to know what OS and version we're running, so the output of
```
cat /etc/os-release
head /boot/*emon*
```
will be helpful.

Secondly, I would really expect most of those packages to be installed by the OS, can we see the output of 
```
pip3 list --user -v
```
and
```
pip3 list -v
dpkg -l python\* | grep ^ii
```
If we really suspect there's a bug in requests/urllib3 then the first step will be to upgrade the OS. We should be using Debian 10 (buster), if this is even an emonpi... Is it one of the published images? Are the system packages up to date?

Also, it's worth reiterating that we don't have any way of testing anything we tinker with, but we do run the risk of breaking other parts of the system. Plus, I have proposed a hopefully harmless change which AFAICT has not been merged, so there is little point in making more changes until then.

borpin, 2020-09-01:
That, in updating the interfacer to Python3, a bug has been introduced that causes the interfacer to stop is, I suggest, beyond doubt - too many independent reports of the issue.

If it definitely isn't the code itself, there must be some other factor in play. My suggestion is simply that we have to look at the environment it is running in.

lgheorghe, 2020-09-03:
@bwduncan - the requested outputs:

```
pi@emonpi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@emonpi:~ $ 
pi@emonpi:~ $ head /boot/*emon*
==> /boot/emonSD-17Oct19 <==

==> /boot/emonsdinit <==
```

```
pi@emonpi:~ $ pip3 list --user -v
Package   Version Location                                    Installer
--------- ------- ------------------------------------------- ---------
redis     3.5.3   /home/pi/.local/lib/python3.7/site-packages pip      
xmltodict 0.12.0  /home/pi/.local/lib/python3.7/site-packages pip      
```

```
pi@emonpi:~ $ pip3 list -v
Package          Version   Location                                    Installer
---------------- --------- ------------------------------------------- ---------
asn1crypto       0.24.0    /usr/lib/python3/dist-packages                       
attrs            18.2.0    /usr/lib/python3/dist-packages                       
Automat          0.6.0     /usr/lib/python3/dist-packages                       
certifi          2018.8.24 /usr/lib/python3/dist-packages                       
chardet          3.0.4     /usr/lib/python3/dist-packages                       
Click            7.0       /usr/lib/python3/dist-packages                       
colorama         0.3.7     /usr/lib/python3/dist-packages                       
colorzero        1.1       /usr/lib/python3/dist-packages                       
configobj        5.0.6     /usr/lib/python3/dist-packages                       
constantly       15.1.0    /usr/lib/python3/dist-packages                       
cryptography     2.6.1     /usr/lib/python3/dist-packages                       
entrypoints      0.3       /usr/lib/python3/dist-packages                       
gpiozero         1.5.1     /usr/lib/python3/dist-packages                       
hyperlink        17.3.1    /usr/lib/python3/dist-packages                       
idna             2.6       /usr/lib/python3/dist-packages                       
incremental      16.10.1   /usr/lib/python3/dist-packages                       
keyring          17.1.1    /usr/lib/python3/dist-packages                       
keyrings.alt     3.1.1     /usr/lib/python3/dist-packages                       
paho-mqtt        1.5.0     /usr/local/lib/python3.7/dist-packages      pip      
pip              18.1      /usr/lib/python3/dist-packages                       
pyasn1           0.4.2     /usr/lib/python3/dist-packages                       
pyasn1-modules   0.2.1     /usr/lib/python3/dist-packages                       
pycrypto         2.6.1     /usr/lib/python3/dist-packages                       
PyGObject        3.30.4    /usr/lib/python3/dist-packages                       
pymodbus         2.1.0     /usr/lib/python3/dist-packages                       
pyOpenSSL        19.0.0    /usr/lib/python3/dist-packages                       
pyserial         3.4       /usr/lib/python3/dist-packages                       
pyserial-asyncio 0.4       /usr/lib/python3/dist-packages                       
python-apt       1.8.4.1   /usr/lib/python3/dist-packages                       
pyxdg            0.25      /usr/lib/python3/dist-packages                       
redis            3.5.3     /home/pi/.local/lib/python3.7/site-packages pip      
requests         2.21.0    /usr/lib/python3/dist-packages                       
RPi.GPIO         0.7.0     /usr/lib/python3/dist-packages                       
SecretStorage    2.3.1     /usr/lib/python3/dist-packages                       
service-identity 16.0.0    /usr/lib/python3/dist-packages                       
setuptools       40.8.0    /usr/lib/python3/dist-packages                       
six              1.12.0    /usr/lib/python3/dist-packages                       
[... 82 lines cut ...]
ii  python3-pkg-resources         40.8.0-1          all          Package Discovery and Resource Access using pkg_resources
ii  python3-pyasn1                0.4.2-3           all          ASN.1 library for Python (Python 3 module)
ii  python3-pyasn1-modules        0.2.1-0.2         all          Collection of protocols modules written in ASN.1 language (Python 3)
ii  python3-pymodbus              2.1.0+dfsg-1      all          full Modbus protocol implementation for Python 3
ii  python3-requests              2.21.0-1          all          elegant and simple HTTP library for Python3, built for human beings
ii  python3-rpi.gpio              0.7.0-0.1~bpo10+1 armhf        Module to control Raspberry Pi GPIO channels (Python 3)
ii  python3-secretstorage         2.3.1-2           all          Python module for storing secrets - Python 3.x version
ii  python3-serial                3.4-4             all          pyserial - module encapsulating access for the serial port
ii  python3-serial-asyncio        0.4-1             all          Async I/O extension package for pyserial
ii  python3-service-identity      16.0.0-2          all          Service identity verification for pyOpenSSL (Python 3 module)
ii  python3-setuptools            40.8.0-1          all          Python3 Distutils Enhancements
ii  python3-six                   1.12.0-1          all          Python 2 and 3 compatibility library (Python 3 interface)
[... cut ...]
ii  python3-xdg                   0.25-5            all          Python 3 library to access freedesktop.org standards
ii  python3-zope.interface        4.3.2-1+b2        armhf        Interfaces for Python3
ii  python3.7                     3.7.3-2+deb10u2   armhf        Interactive high-level object-oriented language (version 3.7)
ii  python3.7-dev                 3.7.3-2+deb10u2   armhf        Header files and a static library for Python (v3.7)
ii  python3.7-minimal             3.7.3-2+deb10u2   armhf        Minimal subset of the Python language (version 3.7)

```

closed 2023-02-10 by borpin