openenergymonitor/emonhub #134: emonhub not receiving data, may not be sending data

wrvolz, 2020-10-12:
I just upgraded to Buster and in the process I also upgraded emoncms and emonhub.  It appears that emonhub is not receiving data. I can see the light on the radio receiver blink about every 10 seconds so I know it getting that far. But nothing is coming into the hub. The log is below:

`2020-10-12 02:46:50,235 INFO     MainThread Opening hub...
2020-10-12 02:46:50,238 INFO     MainThread Logging level set to DEBUG
2020-10-12 02:46:50,239 INFO     MainThread Creating EmonHubJeeInterfacer 'RFM2Pi'
2020-10-12 02:46:50,241 DEBUG    MainThread Opening serial port: /dev/ttyAMA0 @ 38400 bits/s
2020-10-12 02:46:52,248 INFO     MainThread RFM2Pi device firmware version: [RF12demo.12]
2020-10-12 02:46:52,249 INFO     MainThread RFM2Pi device current settings:  O i15 g210 @ 433 MHz q1
2020-10-12 02:46:52,251 INFO     MainThread Setting RFM2Pi calibration: 230V (1p)
2020-10-12 02:46:53,253 DEBUG    MainThread Setting RFM2Pi interval: 300
2020-10-12 02:46:53,361 DEBUG    RFM2Pi     RFM2Pi broadcasting time: 02:46
2020-10-12 02:51:53,411 DEBUG    RFM2Pi     RFM2Pi broadcasting time: 02:51
`
The Config file is:

`[reporters]

# This reporter sends data to emonCMS
[[emonCMS]]
#    Type = EmonHubEmoncmsReporter
    Type = EmonHubEmoncmsHTTPInterfacer
    [[[init_settings]]]
    [[[runtimesettings]]]
        pubchannels = ToRFM12,
        subchannels = ToEmonCMS,
        url = https://localhost/emoncms #(defaults to emoncms.org)
        apikey = whatever...
        sendstatus = 1
        senddata = 1
`
I've tried both Types. http and https and with only url and apikey. This is config is from an earlier version of emoncms.  I do not have an 'interfaces' section.

Now it did seem to be getting some data from somewhere. An earlier part of the log had this:

`2020-10-12 01:51:48,165 INFO emonCMS sending: http://localhost/emoncms/input/bulk.json?apikey=E-M-O-N-C-M-S-A-P-I-K-E-Y&data=[[1602480154.0,8,605,0,170,0,12760,3000,3000,3000,3000,3000,3000,1,0,-64],[1602480168.07,8,606,0,161,0,12770,3000,3000,3000,3000,3000,3000,1,0,-63],[1602480173.82,8,612,0,163,0,12760,3000,3000,3000,3000,3000,3000,1,0,-62],[1602480183.65,8,643,0,164,0,12764,3000,3000,3000,3000,3000,3000,1,0,-63],[1602480193.74,8,634,0,164,0,12768,3000,3000,3000,3000,3000,3000,1,0,-63],[1602480199.17,23,154,152,1058,23,1,0,-56],[1602480203.67,8,640,0,172,0,12779,3000,3000,3000,3000,3000,3000,1,0,-62]...
2020-10-12 01:51:48,188 WARNING emonCMS send failure: wanted 'ok' but got 'false'
`
I'm not getting this data any longer. Also note the last warning. And that the apikey is either not filled in or masked for the log. I also tried copying the first entry above and subbed my apikey and it returned 'false'. 

Help? Thanks in advance

alexandrecuer, 2020-10-12:
Do you have the MQTT interfacer activated ? you need something like that in the emonhub conf :
```
[[RFM2Pi]]
    Type = EmonHubJeeInterfacer
    [[[init_settings]]]
        com_port = /dev/ttyAMA0
        com_baud = 38400                        # 9600 for old RFM12Pi
    [[[runtimesettings]]]
        pubchannels = ToEmonCMS,
        subchannels = ToRFM12,

        group = 210
        frequency = 433
        baseid = 5                              # emonPi / emonBase nodeID
        calibration = 230V                      # (UK/EU: 230V, US: 110V)
        quiet = true                            # Disable quite mode (default enabled) to enable RF packet debugging, show packets which fail crc
        # interval =  300                         # Interval to transmit time to emonGLCD (seconds)

[[MQTT]]

    Type = EmonHubMqttInterfacer
    [[[init_settings]]]
        mqtt_host = 127.0.0.1
        mqtt_port = 1883
        mqtt_user = emonpi
        mqtt_passwd = emonpimqtt2016

    [[[runtimesettings]]]
        pubchannels = ToRFM12,
        subchannels = ToEmonCMS,

        # emonhub/rx/10/values format
        # Use with emoncms Nodes module
        node_format_enable = 1
        node_format_basetopic = emonhub/

        # emon/emontx/power1 format - use with Emoncms MQTT input
        # http://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/MQTT.md
        nodevar_format_enable = 1
        nodevar_format_basetopic = emon/
```
normaly the http_interfacer is more for sending datas to a remote server, on the cloud
if you work locally on an emonpi, the radio interfacer received the datas and the MQTT interfacer publish them to MQTT. Then specific services will do the job within emoncms...

best

Alex

wrvolz, 2020-10-12:
Thanks. Getting somewhere now. I'm getting errors in emonhub.log. But 
first I also have a temp/humid sensor. From looking at the nodes in 
emoncms I was getting data on node 8 and 23. Node 8 appears to 
correspond to a emontx v3 which is what's connected to my power panel. 
Node 23 corresponds to a EmonTH v5.

How do I handle both inputs in emonhub.conf?

Data is not getting to emoncms. I suspect the unreliable content is the 
data from EmonTH. Here's the log:

2020-10-12 08:47:09,410 DEBUG    RFM2Pi     195 NEW FRAME : OK 8 116 9 0 
0 116 0 0 0 8 50 184 11 184 11 184 11 184 11 184 11 184 11 1 0 0 0 (-63)
2020-10-12 08:47:09,413 DEBUG    RFM2Pi     195 Timestamp : 
1602506829.4101598
2020-10-12 08:47:09,413 DEBUG    RFM2Pi     195 From Node : 8
2020-10-12 08:47:09,414 DEBUG    RFM2Pi     195    Values : [2420, 0, 
116, 0, 128.08, 300, 300, 300, 300, 300, 300, 1]
2020-10-12 08:47:09,414 DEBUG    RFM2Pi     195      RSSI : -63
2020-10-12 08:47:09,415 DEBUG    RFM2Pi     195 Sent to channel(start)' 
: ToEmonCMS
2020-10-12 08:47:09,416 DEBUG    RFM2Pi     195 Sent to channel(end)' : 
ToEmonCMS
2020-10-12 08:47:09,475 DEBUG    MQTT       Publishing: 
emon/emonTx_3/power1 2420
2020-10-12 08:47:09,477 DEBUG    MQTT       Publishing: 
emon/emonTx_3/power2 0
2020-10-12 08:47:09,479 DEBUG    MQTT       Publishing: 
emon/emonTx_3/power3 116
2020-10-12 08:47:09,481 DEBUG    MQTT       Publishing: 
emon/emonTx_3/power4 0
2020-10-12 08:47:09,482 DEBUG    MQTT       Publishing: 
emon/emonTx_3/Vrms 128.08
2020-10-12 08:47:09,484 DEBUG    MQTT       Publishing: 
emon/emonTx_3/temp1 300
2020-10-12 08:47:09,486 DEBUG    MQTT       Publishing: 
emon/emonTx_3/temp2 300
2020-10-12 08:47:09,487 DEBUG    MQTT       Publishing: 
emon/emonTx_3/temp3 300
2020-10-12 08:47:09,489 DEBUG    MQTT       Publishing: 
emon/emonTx_3/temp4 300
2020-10-12 08:47:09,491 DEBUG    MQTT       Publishing: 
emon/emonTx_3/temp5 300
2020-10-12 08:47:09,493 DEBUG    MQTT       Publishing: 
emon/emonTx_3/temp6 300
2020-10-12 08:47:09,495 DEBUG    MQTT       Publishing: 
emon/emonTx_3/pulse 1
2020-10-12 08:47:09,496 DEBUG    MQTT       Publishing: 
emon/emonTx_3/rssi -63
2020-10-12 08:47:09,498 INFO     MQTT       Publishing: 
emonhub/rx/8/values 2420,0,116,0,128.08,300,300,300,300,300,300,1,-63
2020-10-12 08:47:10,127 DEBUG    RFM2Pi     Discarding RX frame 
'unreliable content'? 189 75 245 235 248 210 47 141 202 95 48 184 208 
209 181 193 75 151 171 249 203 (-92)
2020-10-12 08:47:12,858 DEBUG    RFM2Pi     Discarding RX frame 
'unreliable content'? 129 77 112 9 204 127 239 97 45 137 10 115 234 109 
216 9 194 248 112 32 176 (-87)
2020-10-12 08:47:15,990 DEBUG    RFM2Pi     Discarding RX frame 
'unreliable content'? 177 64 168 90 1 122 207 10 34 113 79 204 147 0 143 
32 130 27 40 226 102 (-89)
2020-10-12 08:47:16,199 DEBUG    RFM2Pi     Discarding RX frame 
'unreliable content'? 132 247 138 195 15 65 28 10 10 190 239 74 89 225 
245 79 239 154 143 97 162 (-92)
2020-10-12 08:47:16,610 DEBUG    RFM2Pi     Discarding RX frame 
'unreliable content'? 138 5 16 12 2 8 92 6 162 22 13 87 29 105 35 49 144 
116 133 1 75 (-89)

-- 
Bill Volz 4 Acacia Ct Black Mountain, NC, 28711 *832*-724-5183

alexandrecuer, 2020-10-12:
The log seems OK : it is receiving and publishing correctly for node 8
you've got the line 
```
emonhub/rx/8/values 2420,0,116,0,128.08,300,300,300,300,300,300,1,-63
```
so you should see something in emoncms (input module) as far as node 8 is concerned....

Maybe check the node configuration for node 23 in case node 8 works and node 23 no

if you do not receive anything neither from node 23 nor node 8, check the services, for example via the admin module...you should see at least emoncms_mqtt and mosquitto running (green light)....if not the problem should be at this (service) level

wrvolz, 2020-10-12:
First how do I set up emonhub.conf to receive data from both emontx and 
emonth? The discarded data I assume is the temperature data.

I'm not receiving any data in emoncms. I'm not sure where you're talking 
about for checking services. I looked on the emoncms page under 
administration and there's nothing for services there.

I did notice that the devices field is empty - are the emontx and emonth 
supposed to be defined?

I have a configuration problem in the emoncms page. I'm getting this 
message at the top of every web page:

*Warning*: scandir(): (errno 2): No such file or directory 
in*/opt/emoncms/modules/demandshaper/demandshaper-module/demandshaper_model.php*on 
line*58*

and in apache2-error.log has this entry:

[Mon Oct 12 09:49:25.979401 2020] [:error] [pid 3451] [client 
192.168.0.201:52209] PHP Warning:  scandir(/demandshaper/devices): 
failed to open dir: No such file or directory in 
/opt/emoncms/modules/demandshaper/demandshaper-module/demandshaper_model.php 
on line 58, referer: http://192.168.0.3/admin/view

and looking at that php file that $this->dir is not set. I have no clue 
how that is set but suspect it's a configuration problem.

Thanks,
**

-- 
Bill Volz 4 Acacia Ct Black Mountain, NC, 28711 *832*-724-5183

alexandrecuer, 2020-10-12:
check the emonhub.conf file to understand how to setup nodes, yu must have an interfacers section and a nodes section : 
cf https://github.com/openenergymonitor/emonhub/blob/master/conf/emonhub.conf

screen copy of what should be the admin page : 
[image]

if your emoncms installation is broken and if you want to check the services status, you can log in your emonpi through ssh

assuming your pi is 192.168.1.2 and you have a linux desktop, launch a shell and type `ssh pi@192.168.1.2`
then enter your password

once logged, you will be able to check the service status

```
systemctl status emoncms_mqtt.service
systemctl status mosquitto.service
```
How old was your installation ? 
**Did you make a backup before upgrading ?**
you have a fresh image here : https://community.openenergymonitor.org/t/emonsd-24jul20-release/15170
if you have a backup, you can try to restore it on a new SD after burning a fresh image...
the fact is that emoncms ecosystem has got a complex build process, so when things are broken, it is sometimes hard to fix

for demandshaper, I don't know as I don't use this module...maybe go to the OEM community forum :
https://community.openenergymonitor.org/

wrvolz, 2020-10-12:
I've looked at that emonhub.conf. Mine looks similar. What I don't see 
is how to set it up if there are two different transmitters sending to 
the radio - a EmonTx and a EmonTh. Do I put another RFM2pi section or 
what? I don't seen any docs on the EmonTH pages on how to set this up.

My admin page looks a lot different. It's more like the old page:

Mqtt isn't running - I'll look into what's going on there. Mosquitto is 
running.

I didn't take a back up, 1) because I don't know how, or 2) emoncms 
wasn't working after the Buster upgrade and/or 3) I'm an idiot (no 
comments from the peanut gallery).

the server page says it's version 9.8.7 | 2017.06.16.

I have an issue in for demandshaper but no response but perhaps I should 
try the community. I do think a lot is the configuration. I log in at 
192.168.0.3/emoncms but when I go to other pages like admin the page is 
192.168.0.3/admin/view. I'd think that it should be 
192.168.0.3/emoncms/admin/view (which give me a blank page).

Thanks,

-- 
Bill Volz 4 Acacia Ct Black Mountain, NC, 28711 *832*-724-5183

alexandrecuer, 2020-10-12:
there is a module called backup
it can be launched in command line
maybe do a git clone and launch the install process
```
cd /opt/emoncms/modules
git clone https://github.com/emoncms/backup
cd backup
./install.sh
```
check the config.cfg file created by the install process to check where the backup will be created and launch the export:

```
./emoncms-export.sh
```
hope your mariadb server is working fine -)

if not there could be another solution : read the card with a linux machine and copy all the /var/opt/emoncms directory, your feeds are inside...
if you don't have a lot of feeds, it is possible to create an sql script to feed a fresh database....

wrvolz, 2020-10-12:
My RPI runs on a USB attached hard drive so use the SD card only for 
boot. Plenty of space.

Since I've already screwed things up making a back up would just be 
backing up wrong stuff so not sure about that.

I'm still trying to figure out how to get emonhub to handle two 
transmitters - emonTx and emonTh. The best I got was this in 
emonhub.conf but there are still errors. I tried adding another 
[runtimesettings] and emonhub didn't like that. This kinds of works. 
What does the 'group' parameter do? should it be different?

This link fails - 
https://github.com/openenergymonitor/emonhub/configuration.md - not found.

I'm also working on getting emoncms_mqtt.service working. I tried to 
reinstall the mosquitto package using the emoncms script and I'm find 
this -WARNING: Module mosquitto ini file doesn't exist under 
/etc/php/7.0/mods-available
WARNING: Module mosquitto ini file doesn't exist under 
/etc/php/7.0/mods-available

#######################################################################
[interfacers]

# This interfacer manages the RFM2Pi module
[[RFM2Pi]]
     Type = EmonHubJeeInterfacer
     [[[init_settings]]]
         com_port = /dev/ttyAMA0
         com_baud = 38400
     [[[runtimesettings]]]
         pubchannels = ToEmonCMS,
         subchannels = toRFM12,
         group = 210
         frequency = 433
         baseid = 8
         #interval = 300
         quiet = false
         calibration = 110V
[[RFM2Pia]]
     Type = EmonHubJeeInterfacer
     [[[init_settings]]]
         com_port = /dev/ttyAMA0
         com_baud = 38400
     [[[runtimesettings]]]
         pubchannels = ToEmonCMS,
         subchannels = toRFM12,
         group = 210
         frequency = 433
         baseid = 23
         #interval = 300
         quiet = false

-- 
Bill Volz 4 Acacia Ct Black Mountain, NC, 28711 *832*-724-5183

alexandrecuer, 2020-10-13:
Here is the correct link

https://github.com/openenergymonitor/emonhub/blob/master/configuration.md

You should not have two RFM2Pi interfacers déclared in the conf....

Did you tale a look at the scripted install ?
https://github.com/openenergymonitor/EmonScripts

Check the install folder....mosquito script....
https://github.com/openenergymonitor/EmonScripts/blob/master/install/mosquitto.sh

alexandrecuer, 2020-10-13:
> I've asked several times so I'll be clear. I have on RFM module. I have
a EmonTX v3 (node 8). I have a EmonTH v5 (node 23)...... I've seen nothing about how to interface emonhub if 2 or more nodes are sending in data. This seems like a simple question since **OpenEnergyMonitor** sells these devices.

to be clear, the RFM interfacer can manage multiple nodes : there is nothing to do. Just declare one interfacer and the 2 nodes in the conf, that's it. emonhub should do the job....if it does not, it is surely because your emoncms_mqtt service is not working....

wrvolz, 2020-10-13:
OK. I found no examples of that. I tries duplicat RFM and that's not 
right. I tried duplicate runtimessettings and it didn't like that. So 
would it then be baseid = 8,23? Or do I put two baseid's in that section.

I'm in the middle of a complete reinstall of emoncms due to many other 
issue. I made some slight changes to emonhub.conf and now I can't 
restart emonhub. I get this: Fix?

Failed to restart emonhub.service: Unit var-log.mount not found.

-- 
Bill Volz 4 Acacia Ct Black Mountain, NC, 28711 *832*-724-5183

wrvolz, 2020-10-13:
Getting somewhere now. I did a complete removal and reinstall. After a 
few hiccups. I now have the version 10.2.6 running.

Everything is running except emonPiLCD which I don't think it needed for 
standalone installs.

I'm seeing data in emonhub.log for both emontx3 and emonth5. Some log 
from emontx3 and emonth5:

2020-10-13 15:54:41,858 DEBUG    RFM2Pi     248 NEW FRAME : OK 23 205 0 
206 0 44 3 23 0 1 0 0 0 (-58)
2020-10-13 15:54:41,860 DEBUG    RFM2Pi     248 Timestamp : 
1602618881.858024
2020-10-13 15:54:41,861 DEBUG    RFM2Pi     248 From Node : 23
2020-10-13 15:54:41,861 DEBUG    RFM2Pi     248    Values : [20.5, 20.6, 
81.2, 2.3000000000000003, 1]
2020-10-13 15:54:41,862 DEBUG    RFM2Pi     248      RSSI : -58
2020-10-13 15:54:41,862 DEBUG    RFM2Pi     248 Sent to channel(start)' 
: ToEmonCMS
2020-10-13 15:54:41,863 DEBUG    RFM2Pi     248 Sent to channel(end)' : 
ToEmonCMS
2020-10-13 15:54:41,995 DEBUG    MQTT       Publishing: 
emon/emonth5/temperature 20.5
2020-10-13 15:54:41,996 DEBUG    MQTT       Publishing: 
emon/emonth5/external temperature 20.6
2020-10-13 15:54:41,998 DEBUG    MQTT       Publishing: 
emon/emonth5/humidity 81.2
2020-10-13 15:54:42,000 DEBUG    MQTT       Publishing: 
emon/emonth5/battery 2.3000000000000003
2020-10-13 15:54:42,002 DEBUG    MQTT       Publishing: 
emon/emonth5/pulsecount 1
2020-10-13 15:54:42,003 DEBUG    MQTT       Publishing: 
emon/emonth5/rssi -58
2020-10-13 15:54:42,005 INFO     MQTT       Publishing: 
emonhub/rx/23/values 20.5,20.6,81.2,2.3000000000000003,1,-58
2020-10-13 15:54:50,234 DEBUG    RFM2Pi     249 NEW FRAME : OK 8 203 2 0 
0 183 0 0 0 155 49 184 11 184 11 184 11 184 11 184 11 184 11 1 0 0 0 (-60)
2020-10-13 15:54:50,237 DEBUG    RFM2Pi     249 Timestamp : 
1602618890.234576
2020-10-13 15:54:50,238 DEBUG    RFM2Pi     249 From Node : 8
2020-10-13 15:54:50,238 DEBUG    RFM2Pi     249    Values : [715, 0, 
183, 0, 126.99000000000001, 300, 300, 300, 300, 300, 300, 1]
2020-10-13 15:54:50,239 DEBUG    RFM2Pi     249      RSSI : -60
2020-10-13 15:54:50,240 DEBUG    RFM2Pi     249 Sent to channel(start)' 
: ToEmonCMS
2020-10-13 15:54:50,240 DEBUG    RFM2Pi     249 Sent to channel(end)' : 
ToEmonCMS
2020-10-13 15:54:50,418 DEBUG    MQTT       Publishing: 
emon/emontx3/power1 715
2020-10-13 15:54:50,421 DEBUG    MQTT       Publishing: 
emon/emontx3/power2 0
2020-10-13 15:54:50,422 DEBUG    MQTT       Publishing: 
emon/emontx3/power3 183
2020-10-13 15:54:50,423 DEBUG    MQTT       Publishing: 
emon/emontx3/power4 0
2020-10-13 15:54:50,425 DEBUG    MQTT       Publishing: 
emon/emontx3/vrms 126.99000000000001
2020-10-13 15:54:50,426 DEBUG    MQTT       Publishing: 
emon/emontx3/temp1 300
2020-10-13 15:54:50,428 DEBUG    MQTT       Publishing: 
emon/emontx3/temp2 300
2020-10-13 15:54:50,429 DEBUG    MQTT       Publishing: 
emon/emontx3/temp3 300
2020-10-13 15:54:50,431 DEBUG    MQTT       Publishing: 
emon/emontx3/temp4 300
2020-10-13 15:54:50,432 DEBUG    MQTT       Publishing: 
emon/emontx3/temp5 300
2020-10-13 15:54:50,434 DEBUG    MQTT       Publishing: 
emon/emontx3/temp6 300
2020-10-13 15:54:50,435 DEBUG    MQTT       Publishing: emon/emontx3/pulse 1
2020-10-13 15:54:50,437 DEBUG    MQTT       Publishing: 
emon/emontx3/rssi -60
2020-10-13 15:54:50,438 INFO     MQTT       Publishing: 
emonhub/rx/8/values 
715,0,183,0,126.99000000000001,300,300,300,300,300,300,1,-60

However I'm not getting any data in emoncms. If I go to the Feeds page I 
get the dreaded infinite spinning circle of death.

What do I need to do to fix the feeds?

Thanks

-- 
Bill Volz 4 Acacia Ct Black Mountain, NC, 28711 *832*-724-5183

wrvolz, 2020-10-13:
I fixed the feed issue and spinning my wheels. The library location 
changed I copied the data from the previous location to 
/var/opt/emoncms/* and the feed and inputs are now defined.

However not data is appearing. The right side of feeds shows "NULL  
n/a". And if that's not updating neither are the inputs.

I had used prefix='emoncms' in redis so I changed that. At that time the 
last values that were sent showed - about 2.1 days ago.

But it's still not updating in Emoncms with new values. I tried changing 
the basetopic in MQTT section of the emonhub.conf file to both emoncms/ 
and just / and neither did anything. Still not getting current data in 
emoncms. but I can see the old data.

ideas?

Thanks,

-- 
Bill Volz 4 Acacia Ct Black Mountain, NC, 28711 *832*-724-5183