openenergymonitor/emonhub #141: Datacode checks fail on 64-bit non-Windows hosts

cooperised, 2020-12-30:
[Commit e0033560aad1bd6fc5cc2ac2ddd247a6f300e9a6](https://github.com/openenergymonitor/emonhub/commit/e0033560aad1bd6fc5cc2ac2ddd247a6f300e9a6) simplified emonhub_coder.py by removing the hard-coded dictionary of datacodes and sizes and using `struct.calcsize()` instead.

Unfortunately, the datacode "L" (which refers to a 4-byte unsigned field in emonHub) does not return '4' from `struct.calcsize()` on all platforms.  Most 64-bit systems (at least ARM64 and x86_64 with the exception of Windows) define `long` as an 8-byte integer, and on these platforms `struct.calcsize("L")` returns 8.  I'm aware that this does not currently affect the target devices for emonHub, but it may do so in the future.  I came across it because my current interest is in deploying emonHub in a Docker container, and while the eventual host will be a Raspberry Pi I'm doing the dev work on an x86_64 Linux box.

I'd suggest reverting this one part of this commit as a fix.

pb66, 2021-01-05:
I agree, the original code was there to ensure sizes do not vary across os's or architecture, the list was derived from the struct documentation since that is the command used. (see https://docs.python.org/3/library/struct.html#format-characters)

Format | C Type | Python type | Standard size | Notes
-- | -- | -- | -- | --
x | pad byte | no value |   |  
c | char | bytes of length 1 | 1 |  
b | signed char | integer | 1 | (1), (2)
B | unsigned char | integer | 1 | (2)
? | _Bool | bool | 1 | (1)
h | short | integer | 2 | (2)
H | unsigned short | integer | 2 | (2)
i | int | integer | 4 | (2)
I | unsigned int | integer | 4 | (2)
l | long | integer | 4 | (2)
L | unsigned long | integer | 4 | (2)
q | long long | integer | 8 | (2)
Q | unsigned long long | integer | 8 | (2)
n | ssize_t | integer |   | (3)
N | size_t | integer |   | (3)
e | (6) | float | 2 | (4)
f | float | float | 4 | (4)
d | double | float | 8 | (4)
s | char[] | bytes |   |  
p | char[] | bytes |   |  
P | void * | integer |   | (5)

bwduncan, 2021-01-05:
Hi!

Thanks for the feedback.

> The ‘Standard size’ column refers to the size of the packed value in bytes when using standard size; that is, when the format string starts with one of '<', '>', '!' or '='. When using native size, the size of the packed value is platform-dependent.

It's true that `struct.calcsize("L")` can return `8` but we never use native sizes. Every use of struct in this project specifies `little-endian & standard sizes used` (line 16) via the `'<'` prefix, so we will always use the portable sizes. If you want a size of 8, use the `Q` format.

On all platforms `struct.calcsize("<L")` returns `4`.

I'd suggest not reverting this change.

Bruce

cooperised, 2021-01-05:
"We never use native sizes" was not strictly speaking true but is fixed by your pull request I think!  The suggested patch is also consistent with the behaviour of decode() so it's preferable to my suggested reversion.  I'll test tonight after work on a couple of systems and report back.

cooperised, 2021-01-05:
My (non-thorough) testing suggests that this PR fixes the issue.  Thanks!

bwduncan, 2021-01-05:
Thank you. Testing is very highly appreciated :grin:

PR #143 (2021-04-16): Ensure standard sizes
Fixes #141
commit 3fc7f0be (2021-04-16): Ensure standard sizes

Fixes #141
closed 2021-04-16 by TrystanLea