openenergymonitor/emonhub #142: Update the overlaying of the serial ports

borpin, 2021-01-05:
https://github.com/openenergymonitor/emonhub/blob/317f86ec31a995703d86e4bfce818aac75b63f13/install.sh#L24

Quote from Post https://community.openenergymonitor.org/t/emoncms-emonhub-and-bluetooth/16422/21?u=borpin by @pb66 

using `dtoverlay=pi3-disable-bt` does what is says on the tin!

using `dtoverlay=pi3-miniuart-bt` would be a small step in the right direction as it would enable the BT albeit using the lesser serial port.

The correct approach would be to do nothing, leave it as it was, delete that whole block. It would mean changing the addres used in emonhub.conf to `serial0` but that should be done regardless as the GPIO serial port (serial0) is always used regardless of model Pi, regardless of whether BT is onboard/used, regardless of how the serial ports are configured and regardless of which overlay is used(or not).

All that is needed is “enable_uart=1” added to the end of the config.txt to ensure the ttyAMA0 port is enabled depite the console being disabled (by removing “console=serial0,115200” from cmdline.txt of course). The prescience of “enable_uart=1” in config.txt disables the “turbo boosting” functions of the MCU which was the cause of the inconsistent serial baud rates on ttyS0 when the Pi3 was first launched, as it was linked to a variable clock speed. That has not been an issue for years now, we do not need any clock speed adjustments, serial port swaps, BT disabling etc.

[end quote]

The install script should be amended as follows;

Remove `sudo sed -i "s/console=serial0,115200 //" /boot/cmdline.txt`
Remove `sudo sed -i -n '/dtoverlay=pi3-disable-bt/!p;$a dtoverlay=pi3-disable-bt' /boot/config.txt`

Add `enable_uart=1` to `config.txt`
Modify `emonhub.conf` to use `serial0` rather than `ttyAMA0` for serial communications.

This should only be done if emonhub is not already installed (i.e. a fresh install).

Is this correct @pb66? Anything else? (move creating the emonhub user - different issue).

pb66, 2021-01-05:
Remove sudo sed -i "s/console=serial0,115200 //" /boot/cmdline.txt

No that part is needed. (note from the quoted text- ". . . console being disabled (by removing “console=serial0,115200” from cmdline.txt of course).")

    Add enable_uart=1 to config.txt

comments are good, I have this in my scripts

    # Enable GPIO UART as a serial port (but not used for serial console)
    printf '\n# Enable GPIO UART (without console)\nenable_uart=1' | sudo tee -a /boot/config.txt > /dev/null

which results in the following added to the end of config.txt
```
    
    # Enable GPIO UART (without console)
    enable_uart=1
```
note the spacer line so this isn't just tagged directly on to the last line

    Modify emonhub.conf to use serial0 rather than ttyAMA0 for serial communications.

this should really be it's own issue as it's needed regardless, using ttyAMA0 in emonhub.conf is just plain wrong.

Also remove the BT service disable lines if the BT side is reinstated. 

    # We also need to stop the Bluetooth modem trying to use UART
    sudo systemctl disable hciuart

Actually, the current code could be an issue on non-BT Pi's since there is no hciuart service to disable, rather than add tests for the service or for model no etc to determine if BT fitted in order to disable, it would be easier on the installer to just not disable it, leaving it to work on BT Pi's and not if non-BT Pi.

borpin, 2021-01-05:
> No that part is needed. (note from the quoted text- ". . . console being disabled (by removing “console=serial0,115200” from cmdline.txt of course).")

I couldn't work out whether you meant it needed to be removed or not. 😆 Makes sense now.

So this would leave the GPIO UART - Serial0 for the EmonBase/EmonPi and leave the BT active on Serial1.

Correct?  If so sounds like a good move.

closed 2023-02-14 by borpin