openenergymonitor/emonhub #144: http interfacer "UnboundLocalError: local variable 'reply' referenced before assignment"

pb66, 2021-01-14:
See https://community.openenergymonitor.org/t/emonhub-died/16489?u=pb66

Recommend futher indenting line 258 of emonhub_interfacer.py so it becomes part of the Exception and change `reply` to a fixed value or error code etc
https://github.com/openenergymonitor/emonhub/blob/ca13484ac28eeb46da168e48f05004e01d2c6290/src/emonhub_interfacer.py#L249-L258

also there probably needs to be another level of exception unless line 257 will **definitely** catch all possible exceptions.

And, potentially more important, why is this "http" code in the core interfacer code rather than the http interfacer?

borpin, 2021-01-15:
I suppose this come back to what causes emonhub to buffer data.

https://github.com/openenergymonitor/emonhub/blob/ca13484ac28eeb46da168e48f05004e01d2c6290/src/interfacers/EmonHubEmoncmsHTTPInterfacer.py#L72-L72

The HTTP interfacer is looking for `ok` so the reply can be anything other than that.

I note the HTTPInterfacer also uses this `_send_post()` function here

https://github.com/openenergymonitor/emonhub/blob/ca13484ac28eeb46da168e48f05004e01d2c6290/src/interfacers/EmonHubEmoncmsHTTPInterfacer.py#L91-L91

But never checks the return code.

No other interfacer uses it (no surprise).

The suggestion is that this function is moved to the HTTP interfacer (as well as the reply being fixed).

borpin, 2021-01-15:
@TrystanLea, I suggest this is a fairly critical issue. I am surprised it has not come up more often.

borpin, 2021-01-16:
Ref #89 Obvious now what the problem was.

TrystanLea, 2021-01-22:
Thanks for the info on the issue above. Here's my fix so far, suggestions welcome:
https://github.com/openenergymonitor/emonhub/commit/6f94ac38aeb90b2501e75fe18ddff6bf6662f847

- moved the request to the http interfacer, keeping only the POST mode.
- moved the apikey into the post body
- apikey placeholder is hidden in the log (not sure if thats desirable)

Example of the log:

```
2021-01-22 19:47:10,540 DEBUG    MBUS       Decoded MBUS data: {"Energy": [0, "kWh"], "Power": [0, "W"], "Power_Max": [1888, "W"], "Energy5": [0, "kWh"], "Energy6": [0, "kWh"]}
2021-01-22 19:47:10,542 DEBUG    MBUS       19 NEW FRAME : 
2021-01-22 19:47:10,544 DEBUG    MBUS       19 Timestamp : 1611344830.025949
2021-01-22 19:47:10,547 DEBUG    MBUS       19 From Node : MBUS
2021-01-22 19:47:10,549 DEBUG    MBUS       19    Values : [0, 0, 1888, 0, 0]
2021-01-22 19:47:10,550 DEBUG    MBUS       19 Sent to channel(start)' : ToEmonCMS
2021-01-22 19:47:10,553 DEBUG    MBUS       19 Sent to channel(end)' : ToEmonCMS
2021-01-22 19:47:14,690 DEBUG    emoncmsorg Buffer size: 1
2021-01-22 19:47:14,691 INFO     emoncmsorg sending: http://emonpi.local/input/bulk.json? data=[[1611344830.0259485,"MBUS",0,0,1888,0,0]]&sentat=1611344834
2021-01-22 19:47:14,819 DEBUG    emoncmsorg acknowledged receipt with 'ok' from http://emonpi.local

```

TrystanLea, 2021-01-22:
Or with the apikey placeholder printed?

```
2021-01-22 19:55:10,603 DEBUG    MBUS       Decoded MBUS data: {"Energy": [0, "kWh"], "Power": [0, "W"], "Power_Max": [1888, "W"], "Energy5": [0, "kWh"], "Energy6": [0, "kWh"]}
2021-01-22 19:55:10,605 DEBUG    MBUS       1 NEW FRAME : 
2021-01-22 19:55:10,607 DEBUG    MBUS       1 Timestamp : 1611345310.089382
2021-01-22 19:55:10,608 DEBUG    MBUS       1 From Node : MBUS
2021-01-22 19:55:10,608 DEBUG    MBUS       1    Values : [0, 0, 1888, 0, 0]
2021-01-22 19:55:10,609 DEBUG    MBUS       1 Sent to channel(start)' : ToEmonCMS
2021-01-22 19:55:10,610 DEBUG    MBUS       1 Sent to channel(end)' : ToEmonCMS
2021-01-22 19:55:10,892 DEBUG    emoncmsorg Buffer size: 1
2021-01-22 19:55:10,893 INFO     emoncmsorg sending: http://emonpi.local/input/bulk.json? data=[[1611345310.0893824,"MBUS",0,0,1888,0,0]]&sentat=1611345310&apikey=E-M-O-N-C-M-S-A-P-I-K-E-Y
2021-01-22 19:55:10,973 DEBUG    emoncmsorg acknowledged receipt with 'ok' from http://emonpi.local

```

TrystanLea, 2021-01-26:
I've merged this into the master branch for further testing, seems to be working fine for me.

TrystanLea, 2021-01-26:
I've merged the second option with the apikey=E-M-O-N-C-M-S-A-P-I-K-E-Y placeholder shown

closed 2021-01-26 by TrystanLea

borpin, 2021-01-26:
@TrystanLea - you have not removed the code from `emonhub_interfacer.py`

TrystanLea, 2021-02-02:
Sorry for the delay on this, yes removing the phone line from a BT Router results in the following error printed in the log:

> WARNING  emoncmsorg emoncmsorg couldn't send to server: HTTPSConnectionPool(host='emoncms.org', port=443): Max retries exceeded with url: /input/bulk.json (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')])")))

The data is buffered and posted successfully when the connection is re-established

borpin, 2021-02-03:
> you have not removed the code from `emonhub_interfacer.py`

I still see the HTTP code in the main interfacer. I think this can be removed along with the import statement.