openenergymonitor/emonhub #171: install on brand new rpi instance

paultanner, 2022-04-03:
at the end of the install process I get
```
Failed to restart emonhub.service: Unit var-log.mount not found.
```
Presumably this is because it would be normal to have a separate log device named this.
Ideally this would be a warning rather than preventing execution.

I do have a USB drive for logs at /mnt/usb1/log - not sure how to fix the config accordingly as I don't see a reference to var-log.mount anywhere.

paultanner, 2022-05-09:
returned to this and tried creating a symlink from the usb drive to var-log.mount.  same problem.  Double checked the config page but, as yet, there is nothing there about logging.

TrystanLea, 2022-05-10:
Hello @paultanner I wonder if you can try removing the var-log.mount requirement from the service file? It appears to be defined on the following lines: https://github.com/openenergymonitor/emonhub/blob/master/service/emonhub.service#L5

You may be able to just remove the reference to get this working...

TrystanLea, 2022-05-10:
It's also possible to check if emonhub will run and work using the following CLI:

    python3 /opt/openenergymonitor/emonhub/src/emonhub.py --config-file=/etc/emonhub/emonhub.conf

paultanner, 2022-05-11:
Thx Trystan,
Done and it works.
Not sure why that was there.
Now I just need to edit --logfile to put that on a drive other than the default.

closed 2022-05-11 by paultanner