openenergymonitor/emonhub #172: modbus to USB adaptor installation

paultanner, 2022-05-11:
I have followed the config guidance and tried this
```
[[MBUS]]
    Type = EmonHubMBUSInterfacer
    [[[init_settings]]]
        device = /dev/ttyAMA0
        baud = 2400
    [[[runtimesettings]]]
        pubchannels = ToEmonCMS,
        read_interval = 10
        validate_checksum = False
        nodename = MBUS
        [[[[meters]]]]
            [[[[[sontex]]]]]
                address = 1
                type = sontex531
```
The device (from emon shop) is plugged into a USB port on a pi and the A port of this is connected to the MBUS port of my Sontex using a piece of telephone wire.

The error I get is
```
MainThread Device communication error - check settings                         
```
There may be some other config to do, maybe of the Sontex?
Any suggestions please.

glynhudson, 2022-05-13:
Hi Paul, have you checked to see see what mbus address the sontex is set to? Trystan has made a script to check the address and change the address if needed: https://github.com/emoncms/usefulscripts/tree/master/mbus

paultanner, 2022-05-14:
@glyn sent me this https://github.com/emoncms/usefulscripts/tree/master/mbus and I also saw that I was using a serial port not /dev/ttyUSB0.
I modified the check script to try all 254 possible addresses but the meter did not respond on any of them.
Any further suggestions?

glyn, 2022-05-14:
> @glyn sent me this https://github.com/emoncms/usefulscripts/tree/master/mbus and I also saw that I was using a serial port not /dev/ttyUSB0. I modified the check script to try all 254 possible addresses but the meter did not respond on any of them. Any further suggestions?

Did you mean @glynhudson ?

paultanner, 2022-05-14:
Yes @glynhudson.  Forgot to check before submitting.
I have since found that the sontex is set to 2400 baud and address 0.
Am trying to find out how to change that as the sontex doc is unclear.

glynhudson, 2022-05-14:
You can change the Sontex Mbus address using this script: https://github.com/emoncms/usefulscripts/blob/master/mbus/mbus_set_address.py

paultanner, 2022-05-17:
The problem is that the meter does not respond at all.  The device address is set to zero.  I have asked Sontex if I can change that setting another way but have yet to hear back from them.

paultanner, 2022-05-18:
I have now set the meter address to 1. Still no response.

glynhudson, 2022-05-18:
It sounds like you're doing everything correct! Mbus on Sontex has just worked for me assuming the address was set correctly. I'll ask @TrystanLea to take a look, he's the mbus expert, and he wrote the above scripts to set the address and read from the meter.

paultanner, 2022-05-19:
I'm wondering what other settings might be needed?  baud rate looks OK (2400). pi is connected on /dev/ttyUSB0.

paultanner, 2022-05-20:
Yes. Stockshed (Sontex dist) confirm that power is required.  Now I need to know how to do that.  I can run AC power to it but do not have the necessary info to hook it up.  I gather that m-bus needs +/- 15v to operate. Sounds like I should have got the mains-powered version as that might have this other supply rail?

paultanner, 2022-05-20:
Correction: apparently the normal thing is for the data logger to power the m-bus, not the meter (as there may be multiple data sources on the m-bus).  Looks like I need to adapt the m-bus to usb to add the power circuit.  Any suggestions on that?  I'm happy to do some breadboarding/ soldering.

TrystanLea, 2022-05-20:
Hello Paul. The MBUS reader already has an integrated power supply for the MBUS part, but it doesn't power the heat meter. That requires a separate AC power supply for the Sontex itself.

paultanner, 2022-05-20:
Thx Trystan. When I got the Sontex there was a choice of battery or mains power.  I chose battery and the meter is working.  Are you saying that the mains option powers the m-bus whereas the battery does not?

paultanner, 2022-05-20:
Also, when I look at your m-bus to usb adaptor I don't see the kind of components that would supply the voltage for the m-bus.  So I'm wondering how you test it.  If this involves a heat meter then which model exactly.

TrystanLea, 2022-05-20:
Hello Paul, do you have this part?

[image]

The small inductor L1, chip with 5 pins, diode and other components form a boost converter producing +34V. You should be able to measure 32-34V across the capacitor nearest the AA terminals (between A and Ground). Let me know what voltage you see and what voltage you see across A-B as well?

I test them in the office against a Qalcosonic E3, just a heat meter that I have free here.

paultanner, 2022-05-21:
Yes, that's the board I have. I get 32.8v between A and GND.  31.3v between A and B.  The cable is only about 2 m. Seems like lack of power is not the problem.

In mbus_check_address.py there are no checks other than sending the prompt and receiving nothing back.  I'm wondering what else I could do to find the problem?

TrystanLea, 2022-05-23:
Hello Paul, Im a little confused on the power supply issue/requirement for the Sontex531. I remember that being a requirement in the past but there's nothing in the installation manual that suggests it should not work on battery power. It might be worth us sending you another MBUS reader to try, it is quite easy to damage the reader if the A-B terminals are shorted.

paultanner, 2022-05-23:
Thx Trystan,
I've been wondering if I could devise something akin to a loopback test.
Maybe just send frames continuously to make sure the sender part is working.  Maybe then use a switch and resistor test the receiver part.  Could be a good idea before shipping another board, especially as we can't be sure that the meter's m-bus feature is working.  Another idea would be to get hold of a cheap m-bus slave.

glynhudson, 2022-06-12:
Hi Paul, any progresses? We're more than happy to send a replacement reader to rule out any issues with the reader. Please send an email to support@openenergymonitor.zendesk.com to arrange this.