openenergymonitor/emonhub #180: EmonModbusTcpInterfacer Docs missing details

g0lgs, 2022-10-27:
The Docs / Examples for EmonModbusTcpInterfacer appear to be missing the information of how to set the Modbus unit ID.

From what I can find in an old forum thread you need to do:

```
   [[[runtimesettings]]]
        register = 305, 306, 314, 321, 0xA001
        # unit number for each Register
        nUnit = 2, 2, 2, 2, 2

```

This seems to have worked.

alexandrecuer, 2022-10-27:
the doc is here

https://github.com/openenergymonitor/emonhub/tree/master/conf/interfacer_examples/modbus

g0lgs, 2022-10-27:
The Doc might be, but it does not cover the setting I amd refering too.

glynhudson, 2022-10-27:
What device are you reading data from? The modbus ID needs to be set on the device

g0lgs, 2022-10-28:
I have the Device ID set to 2.

The Docs / Examples for the Interfacer do not appear to show how to configure the ID to match and defaults to 1