openenergymonitor/emonhub #197: Relative path for available.conf?

glynhudson, 2023-02-24:
Could we use a relative path here i.e `conf/available.conf` instead of hardcoding the path to handle emonhub being installed in a different location? 

https://github.com/openenergymonitor/emonhub/blob/f67244500821b196f98c54f15062bcc60a757daf/src/emonhub_auto_conf.py#L49

h1771t, 2023-03-11:
please also see: https://community.openenergymonitor.org/t/mqtt-interfacer-gets-no-data/23097
related to this issue when trying to use emonhub as standalone.
Thanks

borpin, 2024-02-16:
Could create a symlink from `/etc/emonhub/` at install time to point back to where `autoconf` is and use the `/etc/emonhub/` location in the script.

borpin, 2024-02-29:
Another instance

https://community.openenergymonitor.org/t/power-outage-and-suddenly-stops-sending-data-to-emoncms/25611/22?u=borpin