openenergymonitor/emonhub #213: Change Endian.Big to Endian.BIG as per pymodbus change

jkjaer, 2024-05-14:
In https://github.com/pymodbus-dev/pymodbus/pull/1743, pymodbus changed constants.Endian into being an enumeration, resulting in that Endian.Big was changed into Endian.BIG. Unfortunately, this change has broken the  EmonModbusTcpInterfacer class which uses Endian.Big for both the byte and word order when calling BinaryPayloadDecoder.fromRegisters (also from pymodbus).

To fix the problem, Endian.Big should be changed to Endian.BIG in EmonModbusTcpInterfacer.py.

jkjaer, 2024-05-14:
I have already fixed the problem on a new branch on a local copy of the emonhub repository. However, I do not have access rights to push this fix to here.

alexandrecuer, 2024-05-16:
@jkjaer : the procedure is to make a pull request, you fork the repo into your github account, then make the mods and then open a PR.

there is also other small things to correct : 
`read_holding_registers(register - 1, qty, unit=unitId)` should be `read_holding_registers(register - 1, qty, slave=unitId)`
since pymodbus3

cf https://github.com/openenergymonitor/emonhub/issues/182

tell me if you are OK for the PR, otherwise I will do it

jkjaer, 2024-05-17:
@alexandrecuer Thanks for the clarification. I will implement the two fixes and open a PR.

alexandrecuer, 2024-05-18:
@jkjaer : thanks for the PR 👍

jkjaer, 2024-05-18:
No problem. Glad to contribute :)

closed 2024-05-18 by jkjaer