openenergymonitor/emonhub #216: emonhub.conf files reference stale/old branch configuration documentation

reedy, 2024-06-16:
https://github.com/openenergymonitor/emonhub/blob/f4fd946/conf/default.emonhub.conf
https://github.com/openenergymonitor/emonhub/blob/34ce506/conf/emonhub.conf
https://github.com/openenergymonitor/emonhub/blob/51316db/conf/emonpi.default.emonhub.conf
https://github.com/openenergymonitor/emonhub/blob/5b734da/conf/emonpi2.default.emonhub.conf
https://github.com/openenergymonitor/emonhub/blob/51316db/conf/example.emonhub.conf

```
### emonHub configuration file, for info see documentation:
### https://github.com/openenergymonitor/emonhub/blob/emon-pi/configuration.md
```

https://github.com/openenergymonitor/emonhub/blob/emon-pi/configuration.md links to the file as of e471f7dea9d99b927c350726eb5567b2a6b38c73, from 4 years ago

Which is somewhat different to https://github.com/openenergymonitor/emonhub/blob/master/docs/configuration.md

This presumably means, even on new(er) installs, the config files are being created by default, referencing oooold documentation

TrystanLea, 2024-06-16:
Yes agreed would be good to clear this up!

reedy, 2024-06-16:
While poking around other stuff, the url is similarly pointing to the branch further down too

```
## See config user guide: https://github.com/openenergymonitor/emonhub/blob/emon-pi/configuration.md
```

closed 2024-08-16 by glynhudson